carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #14: Use Pin 5 for SD Chip select does not work.

FreddyVRetro, 2023-11-12:
Hi,

Thanks a lot for your code, I use it in the PicoMEM Board.
Il the previouis Board I di redo a PIO driver because 2 of my PIN (TX/RX) Were reversed, classical.
But this did work on a few SD only.

Now, I try with this code and I have problem:

I want to use Pin5 for CS, but it does not move.
In fact it move, but before we do mount.

After initialisation, I have one uSD Working, With CS Always down. (Strange it is working)

Do you use Pin5 for something else somewhere ?

FreddyVRetro, 2023-11-12:
I used the wrong GPIO... Pin 5 is GPIO 3

It is working better, on more SD than before.

I have a Kingston that fail : 
sd_spi_go_low_frequency: Actual frequency: 398089
V2-Version Card
R3/R7: 0x100
CMD8 Pattern mismatch 0x1aa : 0x100

carlk3, 2023-11-12:
I can't think of any use of Pin5 in particular. Someone else had a problem with pin 25, due to that being the PICO_DEFAULT_LED_PIN and some code in no-OS-FatFS-SD-SDIO-SPI-RPi-Pico\src\sd_driver\SPI\spi.h that used the LED as a drive activity indicator, but I have disabled that code by default. But, you might see what pins are used in pico-sdk\src\boards\include\boards\pico.h.

carlk3, 2023-11-12:
> ...
> I have a PNY 32Gb that fail : sd_spi_go_low_frequency: Actual frequency: 398089 V2-Version Card R3/R7: 0x100 CMD8 Pattern mismatch 0x1aa : 0x100

Are you trying to use that with SPI or SDIO? Be aware that once a card is switched to SPI mode, the only way to get it back to SD mode is to power cycle it.

CMD8 is supposed to determine the voltage range supported by the SD card, but it is an early command and the card might be rejecting it for something else it is unhappy about.  Most often, this is due to something it doesn't like about the Vdd or GND or the CLK.

You're getting R3/R7: `0x100`, which is `0b100000000`, so bit 8 is on.

[image]

Bit 8 says "2.7 - 3.6V":

[image]

so that much is good.

[image]

The "check pattern" was expected to be `0xAA` but you got `0x00`, so the card is unhappy about something. I suspect an electrical problem.

FreddyVRetro, 2023-11-12:
Thanks.

I planned to add more capas on the 3.3v for my next board release.
I did not add pull up as well.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

...

[image]<https://user-images.githubusercontent.com/50121841/282312177-4cb7574c-0eda-4a7a-9a36-1a8ba9263ce4.png>

[image]<https://user-images.githubusercontent.com/50121841/282312352-e560c36b-3ea4-4378-895c-6eb3ed2fdce7.png>

[image]<https://user-images.githubusercontent.com/50121841/282312560-014f2cb1-db18-46b9-89a8-f1ecee1500b3.png>

carlk3, 2023-11-13:
> I did not add pull up as well.

I don't really use pull ups anymore. Modern SD cards all have push-pull outputs, so the old thing about needing one on the card's DO doesn't seem to apply anymore. I think it was, or maybe still is, required for MMC cards, but I never use those. 

These days, I'm more interested in serial source terminating resistors. I find these effective for running at higher speeds while also reducing noise (radiated EMI as well as local noise coupled into the power supply, etc.)

closed 2024-08-20 by carlk3