carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #16: ss_gpio should be optional

bog-dan-ro, 2023-11-26:
It's very useful for projects that are using a single SPI sdcard and the CS is hardwired, this way we save a pin ;-) .

bog-dan-ro, 2023-11-27:
Ah, you're right

closed 2023-11-27 by bog-dan-ro

carlk3, 2023-11-27:
If you want to save pins, I think a better approach is to use 1-bit SDIO. 

[image]
For SPI, you need:

- SCLK; clock
- MOSI (or COPI, or "peripheral's Data Input [DI]")
- MISO (or CIPO, or "peripheral's Data Output [DO]")
- SS (or CS); select

For 1-bit SDIO you need:
- CLK; clock
- CMD; command
- DAT0; data (bidirectional)

There is an implementation in [pico_sd_card](https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sd_card). I am considering adding that functionality to this library.