carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #40: Need help with Waveshare 2.8inch Display for Pi Pico pin configuration of its onboard SD card

dkts2000, 2024-05-20:
Hi
I am trying to understand how to configure the pins on the Waveshare 2.8inch Display Module for Raspberry Pi Pico to access the onboard SD card using your driver (SDIO).

The Pinout Definition of the board is:

[image]

In the /simple_sdio example if i set 
`
    .CMD_gpio = 18,
    .D0_gpio = 19,
`

the `CLK_gpio` will become 17 (`CLK_gpio = D0_gpio -2.`)
whereas it should be 5 (if i understand correctly)

Can you give me a hint on the configuration?
Thanks!

carlk3, 2024-05-21:
Unfortunately, you'll have to modify the PIO code. (The Pico SDK doesn't provide a way to pass in the offset; I wish it did.) 

You need SDIO_CLK offset from D0 by -14, which is **18** in mod32 arithmetic. 
(19 + 18) % 32 = 5

In [rp2040_sdio.pio](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/main/src/sd_driver/SDIO/rp2040_sdio.pio) you'll need to change this line:
[.define PUBLIC SDIO_CLK_PIN_D0_OFFSET 30    ; (-2 in mod32 arithmetic)](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/6bf8a95bd5e51d4a6cd03bd6e063d74de64335ae/src/sd_driver/SDIO/rp2040_sdio.pio#L39C1-L39C71) to something like
```
.define PUBLIC SDIO_CLK_PIN_D0_OFFSET 18    ; (-14 in mod32 arithmetic)
```

dkts2000, 2024-05-21:
Thanks i ll give it a try!

I also tried the /simple example that comes with your library (using spi) and the following pin configuration

```c
/* Configuration of RP2040 hardware SPI object */
static spi_t spi = {  
    .hw_inst = spi0,  // RP2040 SPI component  ------  (i also tried spi1)
    .sck_gpio = 5,    // GPIO number (not Pico pin number)
    .mosi_gpio = 18,
    .miso_gpio = 19,
    .baud_rate = 12 * 1000 * 1000   // Actual frequency: 10416666.
};

/* SPI Interface */
static sd_spi_if_t spi_if = {
    .spi = &spi,  // Pointer to the SPI driving this card
    .ss_gpio = 22      // The SPI slave select GPIO for this SD card
};
``` 

I always get the message "The physical drive cannot work" in a loop, without actually being in a loop in code.

The only way i can get the sd card to initialize and read its contents successfully is by using the driver that Waveshare has in its site that combines sd card and lcd driver. [https://files.waveshare.com/upload/f/fc/Pico-ResTouch-LCD-X_X_Code.zip](url)

I would like to use a separate lcd driver (that gives frame buffer capabilities from [here](https://github.com/tvlad1234/pico-displayDrivs)) and your library to access the sd card but i am stuck.

carlk3, 2024-05-21:
I've never used that board, but didn't you post this for that?
[image]

> ```
> static spi_t spi = {
>     .hw_inst = spi0,  // RP2040 SPI component  ------  (i also tried spi1)
>     .sck_gpio = 5,    // GPIO number (not Pico pin number)
>     .mosi_gpio = 18,
>     .miso_gpio = [...]

I don't understand your choice of GP18 for MOSI and GP19 for MISO, when the table you posted shows GP11 and GP12, respectively. Please explain.

carlk3, 2024-05-21:
Have a look at [Appendix E: Troubleshooting](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico#appendix-e-troubleshooting).  I would start by compiling with `USE_DBG_PRINTF` and cutting the baud rate in half.

carlk3, 2024-05-21:
Also, I think there is a problem with your pin assignments if you want to use SPI0.
[image]

> ```
> /* Configuration of RP2040 hardware SPI object */
> static spi_t spi = {
>     .hw_inst = spi0,  // RP2040 SPI component  ------  (i also tried spi1)
>     .sck_gpio = 5,    // GPIO number (not Pico [...]

- `GP5` is `SPI0 CSn`, not SCK. 
- `GP18` is `SPI0 SCK`, not MOSI.
- `GP19` is `SPI0 TX` (transmit) which would be "Master Out, Slave In" (MOSI), not MISO.

I'm not sure what you could use for MISO (SPI0 RX). Do you have a schematic for the Waveshare board?

dkts2000, 2024-05-21:
I must have messed up every single aspect of it!!! Thanks for the clarifications.

from the pin configuration of waveshare's driver:
```c++
#define LCD_RST_PIN		15	
#define LCD_DC_PIN		8
#define LCD_CS_PIN		9
#define LCD_CLK_PIN		10
#define LCD_BKL_PIN		13
#define LCD_MOSI_PIN	11
#define LCD_MISO_PIN	12
#define TP_CS_PIN		16
#define TP_IRQ_PIN		17
#define SD_CS_PIN		22

#define SPI_PORT		spi1
``` 

i thought that gpios that have LCD_ are for lcd use only

The schematic for this board is [here](https://files.waveshare.com/upload/d/dc/Pico-ResTouch-LCD-2.8_Sch.pdf)

carlk3, 2024-05-21:
It looks like there are some provisions for running the microSD on SPI:
[image]
What position are your jumpers in?

carlk3, 2024-05-21:
Your only choices for SPI0 RX (MISO) are GP4 or GP16. GP16 appears to be assigned to TP_CS. So, you'll need to use GP4 for `miso_gpio`. Somehow, you'll have to route GPIO4 to SD_SDO". Maybe you can pop off the 0Ω resistor on H3 and run a jumper to GPIO4. Or use SDIO instead of SPI.

dkts2000, 2024-05-21:
I 'll check the jumpers later, when i get back.

santolucito, 2024-05-26:
My pico (wavshare 0.96 with adafruit sdio sd readers) was also in a boot loop when trying to connect to the SD cards (I have 2 connected). Setting PICO_XOSC_STARTUP_DELAY_MULTIPLIER to 64 fixed it for me. No idea if it is the same situation for you but might be worth a try.

carlk3, 2024-05-26:
Usually when you're getting CRC errors, the baud rate is too high. As a test, you can set the baud rate quite low; e.g.:
```C
.baud_rate = 1000000
```
I often do this so I can use a $10 logic analyzer to watch the SPI bus. It will be slow, but it should work.

**Note:** The actual baud rate will be the peripheral clock frequency (_CLK_PERI_ ) divided by an even number. For example, 125000000 / 128 = 976562. (See _4.4.2.3. Clock prescaler_ in **RP2040 Datasheet**).

carlk3, 2024-05-26:
> i don't understand why the pico resets continually. 

Nor do I. How do you know it is doing that?

There are 10 retries in [in_sd_go_idle_state](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/6bf8a95bd5e51d4a6cd03bd6e063d74de64335ae/src/sd_driver/SPI/sd_card_spi.c#L936). There are also 3 retries in [sd_cmd](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/6bf8a95bd5e51d4a6cd03bd6e063d74de64335ae/src/sd_driver/SPI/sd_card_spi.c#L460). Could that be what you're referring to? Sometimes it takes a several retries to get a card to Idle state if it was left in some indeterminate state (e.g., if you're running a program under the debugger and restart while the card is in the middle of some operation).

archearth, 2024-06-30:
Hello,

So I have the Pico-ResTouch-LCD-3.5 display I have gotten it o work with my Pico W through SPI1 and with the following PIN setup using the TFT_eSPI library:

#define TFT_MISO 12 
#define TFT_MOSI 11 
#define TFT_SCLK 10 
#define TFT_CS 9
#define TFT_DC 8
#define TFT_RST 15
#define TFT_BL 13

However, I want to be able to connect and use the SD to load images/gifs. I was able to get to load images and gifs from LittleFS but when I try from the SD it renders in a glitchy slow way. I think the I dont have the right SPI/PINS and SD config for the SD and the display to work at the same time. I was able to get the SD to initialize by setting this

SPI1.setSCK(SD_SCLK_PIN); //10
SPI1.setTX(SD_MOSI_PIN); //11
SPI1.setRX(SD_MISO_PIN); //12

But I think there is overlap happening and the display and SD are accessing those at the same time. So I found this library as a potential solution to use the SD card in SDIO mode but not sure how to properly configure the pins. Also when I tried to compile via platformIO I get this error - `src\sdtest.cpp:119:32:` error: 'sd_card_t' {aka 'struct sd_card_t'} has no member named 'state'`

Any idea why and how to configure the SD in SDIO mode?

carlk3, 2024-07-01:
> Any idea why and how to configure the SD in SDIO mode?

In what position are your jumpers on the A B headers? 

[image]

From the schematic, it looks like you need to move them to the B position for SDIO.

archearth, 2024-07-01:
This is how it looks on the back. Also if I have to do that, I am not quite sure how. I am new to all this so trying to figure things while I work on the project.
[image]

carlk3, 2024-07-01:
You're going to have to desolder the zero-ohm resistors in row "A" of H1, H2, H3, H5, and H6 and solder them in row "B".

archearth, 2024-07-01:
Oh I see. I dont think I have the proper tools to handle these small resistors, meaning solder them back to row B. I guess at this point it will be stick to SPI for both the display and SD card and figure our how to configure the PIN so that the can both operate through the SPI or find a different display maybe.

closed 2024-08-20 by carlk3