carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #44: Unused variable warnings in debug code

willeccles, 2024-06-05:
When compiled with `-Wunused-parameter`, the various debug functions result in a lot of warnings. Here's a small snippet, though this is far from complete:

<details>
<summary>Warnings</summary>
```
In file included from /home/REDACTED/lib/no-OS-FatFs/src/include/util.h:20,
                 from /home/REDACTED/lib/no-OS-FatFs/src/sd_driver/sd_regs.h:4,
                 from /home/REDACTED/lib/no-OS-FatFs/src/sd_driver/sd_card.h:36,
                 from /home/REDACTED/lib/no-OS-FatFs/src/include/hw_config.h:18,
                 from /home/REDACTED/lib/no-OS-FatFs/src/src/glue.c:24:
/home/REDACTED/lib/no-OS-FatFs/src/include/my_debug.h: In function 'dump_bytes':
/home/REDACTED/lib/no-OS-FatFs/src/include/my_debug.h:99:51: warning: unused parameter 'bytes' [-Wunused-parameter]
   99 | static inline void dump_bytes(size_t num, uint8_t bytes[]) {
      |                                           ~~~~~~~~^~~~~~~
In file included from /home/REDACTED/lib/no-OS-FatFs/src/src/my_debug.c:24:
/home/REDACTED/lib/no-OS-FatFs/src/include/my_debug.h: In function 'dump_bytes':
/home/REDACTED/lib/no-OS-FatFs/src/include/my_debug.h:99:51: warning: unused parameter 'bytes' [-Wunused-parameter]
   99 | static inline void dump_bytes(size_t num, uint8_t bytes[]) {
      |                                           ~~~~~~~~^~~~~~~
/home/REDACTED/lib/no-OS-FatFs/src/src/my_debug.c: In function 'error_message_printf':
/home/REDACTED/lib/no-OS-FatFs/src/src/my_debug.c:85:60: warning: unused parameter 'func' [-Wunused-parameter]
   85 | int __attribute__((weak)) error_message_printf(const char *func, int line, const char *fmt, ...) {
      |                                                ~~~~~~~~~~~~^~~~
/home/REDACTED/lib/no-OS-FatFs/src/src/my_debug.c:85:70: warning: unused parameter 'line' [-Wunused-parameter]
   85 | int __attribute__((weak)) error_message_printf(const char *func, int line, const char *fmt, ...) {
      |                                                                  ~~~~^~~~
/home/REDACTED/lib/no-OS-FatFs/src/src/my_debug.c: In function 'debug_message_printf':
/home/REDACTED/lib/no-OS-FatFs/src/src/my_debug.c:112:60: warning: unused parameter 'func' [-Wunused-parameter]
  112 | int __attribute__((weak)) debug_message_printf(const char *func, int line, const char *fmt, ...) {
      |                                                ~~~~~~~~~~~~^~~~
/home/REDACTED/lib/no-OS-FatFs/src/src/my_debug.c:112:70: warning: unused parameter 'line' [-Wunused-parameter]
  112 | int __attribute__((weak)) debug_message_printf(const char *func, int line, const char *fmt, ...) {
      |                                                                  ~~~~^~~~
In file included from /home/REDACTED/lib/no-OS-FatFs/src/include/util.h:20,
                 from /home/REDACTED/lib/no-OS-FatFs/src/src/rtc.c:22:
/home/REDACTED/lib/no-OS-FatFs/src/include/my_debug.h: In function 'dump_bytes':
/home/REDACTED/lib/no-OS-FatFs/src/include/my_debug.h:99:51: warning: unused parameter 'bytes' [-Wunused-parameter]
   99 | static inline void dump_bytes(size_t num, uint8_t bytes[]) {
      |                                           ~~~~~~~~^~~~~~~
```
</details>

These should all be fixable by either using the parameters or intentionally ignoring them, e.g., with `(void)bytes;` and similar.

willeccles, 2024-06-05:
I've also just noticed that `sd_sdio_syncDevice()` in `sd_card_sdio.c` also does not use its parameter.

carlk3, 2024-06-06:
OK, I checked in fixes to branch [dev](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev). Please give it a try.

closed 2024-08-20 by carlk3