carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #45: f_mkdir() causes file system errors when mounted back in windows

tgiacchi, 2024-06-05:
[image]

when creating a directory the directory is created, but when going to a command prompt in windows with the SD card mounted,
I will get 'The parameter is incorrect.'   This happens on both exFat and Fat32.  I am very worried about data corruption.

I can run chkdsk /f against the drive and it reports no errors.  Could this be a bug in windows 11?

At 1st, I thought it might be an issue with not unmounting the SD card, but I unmount before ejecting it.  Same results.

carlk3, 2024-06-05:
Weird. Maybe there's something funny about the directory names. What happens if you list the directory with FatFs? E.g., [void ls(const char *dir)](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/cd0b9168272828a3341515028f3496229ed28877/src/src/f_util.c#L110). If you have a Linux box (or WSL) it might be interesting to see what it reports.

How are you formatting the SD card?

tgiacchi, 2024-06-05:
I've done fat32, 16k clusters.  Exfat at different cluster sizes.  File explorer seems to be ok with it.

carlk3, 2024-06-05:
> I've done fat32, 16k clusters.  Exfat at different cluster sizes.  

On Windows? You could also try using FatFs, e.g. [static void run_format(const size_t argc, const char *argv[])](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/cd0b9168272828a3341515028f3496229ed28877/examples/command_line/src/command.cpp#L199), or [SD Memory Card Formatter](https://www.sdcard.org/downloads/formatter/).

> File explorer seems to be ok with it.

So, File Explorer shows it OK but DIR in CMD does not? Are there Unicode characters in the directory names?

tgiacchi, 2024-06-05:
Nope. Just ascii.

tgiacchi, 2024-06-05:
Qq. How do I set the datetime that the code uses

tgiacchi, 2024-06-05:
So the problem stems from not initializing the rtc with some valid value. Cmd doesn't understand all zeros

carlk3, 2024-06-05:
> Qq. How do I set the datetime that the code uses Get Outlook for Android<https://aka.ms/AAb9ysg>

You need to define 
```C
DWORD get_fattime (void);	/* Get current time */
```
E.g., [get_fattime](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/cd0b9168272828a3341515028f3496229ed28877/src/src/rtc.c#L84C7-L84C18).

closed 2024-08-20 by carlk3