carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #50: few small hardware changes for pico sdk 2.0

pauldevine, 2024-08-10:
My c skills are meager, so I was hesitant to offer this as a PR. But I wanted to give you a heads up there's a few small changes to the pico 2.0 SDK hardware definitions that are required to build the library. In case this saves any time for you, I have them here:
[2.0-sdk-differences](https://github.com/pauldevine/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/compare/carlk3%3Ano-OS-FatFS-SD-SDIO-SPI-RPi-Pico%3Amain...main)

carlk3, 2024-08-10:
Thanks! Will the changed code still compile with the earlier SDK, or does there need to be some kind of conditional compilation or something?

sneakywumpus, 2024-08-11:
@carlk3 I was wondering why you directly manipulate the hardware state instead of using the SDK function `gpio_set_slew_rate`, i.e.
```
gpio_set_slew_rate(gpio, GPIO_SLEW_RATE_FAST);
```
instead of
```
padsbank0_hw->io[gpio] |= PADS_BANK0_GPIO0_SLEWFAST_BITS;
```

carlk3, 2024-08-11:
> @carlk3 I was wondering why you directly manipulate the hardware state instead of using the SDK function `gpio_set_slew_rate`, i.e.
> 
> ```
> gpio_set_slew_rate(gpio, GPIO_SLEW_RATE_FAST);
> ```
> 
> instead of
> 
> ```
> padsbank0_hw->io[gpio] |= PADS_BANK0_GPIO0_SLEWFAST_BITS;
> ```

That's ported directly from [ZuluSCSI-firmware](https://github.com/ZuluSCSI/ZuluSCSI-firmware/tree/main): [gpio_conf](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/09a41cc71ed6eaeffc709ef6fa24f3e0714db523/lib/ZuluSCSI_platform_RP2040/ZuluSCSI_platform.cpp#L71). I think there is some value in not making too many changes to the ported code, so that when ZuluSCSI makes improvements to their code I can easily retrofit them (although I am a bit behind in that effort at the moment).

pauldevine, 2024-08-12:
@carlk3 The changes were all due to a change in the underlying SDK. If you look at the [Pico Datasheet](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf) starting on page 628 there's release notes for the 2.0 release. Then on page 634 "The enum gpio_function no longer exists and has been replaced with gpio_function_t." In order to support both 1.5 and 2.0 I believe you will need to do some conditional #ifdef or something to handle the difference. I wasn't intending this to be a PR you'd take as-is, but more an FYI of what places the code needs adjustment to work in the 2.0 SDK. I didn't think my c skills were up to submitting a PR.

carlk3, 2024-08-12:
> In order to support both 1.5 and 2.0 I believe you will need to do some conditional #ifdef or something to handle the difference. 

I guess there's no real need to be backward compatible. I'll just make a new git branch, and the old branch will still be there for anyone on the old SDK.

carlk3, 2024-08-21:
See #53

pauldevine, 2024-09-23:
Excited to see your changes here @carlk3 ! I'm closing this as I see you've successfully migrated to SDK 2.0. I can confirm your changes work for me with SDK 2.0 on a Pimoroni pico plus 2, with the RP2350B.

closed 2024-09-23 by pauldevine