carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #58: Specify RP2350, compile cmmand routine and get error

Octopus1633, 2024-09-09:
Hello，

RP2350 does not have RTC. I specified RP2350 and compiled the cmmand routine, but got an error:
C:\Users\xxx\Desktop\code\Pico-Eval-Board-Code\c\08_SD_CARD\example\src\command.cpp:26:10: fatal error: hardware/rtc.h: No such file or directory

Other errors still appear after commenting. There are many RTC-related codes in the program. Will this example support RP2350 in the future?

arturo182, 2024-09-09:
For context, it looks like the code is from https://www.waveshare.com/wiki/Pico-Eval-Board#Download_Demo and is based on https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/e9df9e4734572183d197fcfe3eaecbf97152f2b3/examples/command_line/src/command.cpp

Octopus1633, 2024-09-09:
Yes that's right, because I want to try to run this example on RP2350.

https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/e9df9e4734572183d197fcfe3eaecbf97152f2b3/examples/command_line/src/command.cpp

carlk3, 2024-09-10:
In the [dev](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev) branch, I have fixed it so that it builds OK with PICO_BOARD pico2. I haven't yet received a pico2, so you'll have to let me know whether it works.

Octopus1633, 2024-09-10:
Hello, thank you for your contribution. I used this branch and successfully compiled and ran it on pico2, but there seems to be some problems in executing the command.

[image]

carlk3, 2024-09-10:
My guess is that you need to run a lower baud rate. There are some other suggestions at [Appendix E: Troubleshooting](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev#appendix-e-troubleshooting).

Octopus1633, 2024-09-12:
I lowered the baud rate and solved the problem, thanks!

closed 2024-09-27 by Octopus1633