carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #63: Tag/Versions newer than 3.1.0?

diegoparrilla, 2024-09-18:
Hi!

I might be overlooking something, but I need help finding a branch or tag newer than 3.1.0, even though the README states that the current version is 3.4.0.

Am I missing something? How can I check out the v3.4.0 version?

Thanks!

leondisctech, 2024-09-18:
+1

Also, it looks like 3.1.0 is a branch rather than a tag or version, which is a little unusual. I think that tags in the main branch would be less confusing.

carlk3, 2024-09-19:
OK, I created Release [v3.4.0](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/releases/tag/v3.4.0). I like to keep the older versions as branches so that I can continue development along one of those older branches if necessary. For example, v2.5.0 dropped support for SD Standard Capacity Memory Cards. If someone is still using those, they can run that branch. If there is some serious bug or something that needs to be rolled back, that branch can be continued.

diegoparrilla, 2024-09-24:
Thank you! Awesome work!