carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #64: simple example build issues, usb device not recognized

gaby64, 2024-09-18:
I am trying to build with vscode and pico extension for the rp2040 pi pico

I was getting the rtc.h not found also, I changed it to my_rtc.h
this builds successfully but when i put the uf2 on the pico, it shows up a popup of unrecognized usb device.

I tried the older repository and that one works fine, writes to sdcard and outputs printf over usb

carlk3, 2024-09-19:
I fixed some bad `#include`s in the [dev](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev) branch.

I don't know what to tell you about the USB device. Maybe review _Chapter 5. Say "Hello World" in C_ in [Getting started with Raspberry Pi Pico-series](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) and _4.5.20.8. pico_stdio_usb_ in [Raspberry Pi Pico-series C/C++ SDK](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf).

> I tried the older repository and that one works fine, writes to sdcard and outputs printf over usb

Which repository is that? Maybe you could Meld or WinMerge or something to compare the projects and look for relevant differences.

gaby64, 2024-09-19:
https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico

carlk3, 2024-09-20:
Did you customize [hw_config.c](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/blob/main/examples/simple/hw_config.c) to match your hardware?

Max585t, 2024-12-16:
I am also experiencing an issue with USB device not being recognized by windows after uploading the simple example uf2

emblink, 2025-05-23:
NOTE: If the _**USB device is not recognized by the host**_ (e.g., Windows),
**Check if you're calling `sleep_ms()` in the main loop**.
Sleeping for too long can prevent TinyUSB from processing USB tasks properly.
Spent over an hour on this, leaving it here to save someone else some time!

carlk3, 2025-05-28:
> NOTE: If the _**USB device is not recognized by the host**_ (e.g., Windows), **Check if you're calling `sleep_ms()` in the main loop**. Sleeping for too long can prevent TinyUSB from processing USB tasks properly. Spent over an hour on this, leaving it here to save someone else some time!

Is that call to `sleep_ms()` in your code?

emblink, 2025-05-28:
> Is that call to `sleep_ms()` in your code?
Yes, in the main() while loop alongside tud_task();