carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #67: Reduce build size

hellcats88, 2024-09-24:
Hi,

I'm compiling for RP2040 pi pico board and my sketch with GxEPD2 and LVGL libraries has a size of 165KB.
When I add your library and only the code
```
```

carlk3, 2024-09-24:
I don't know that the uf2 file size is a good indication. Something like `-Wl,--print-memory-usage` would probably be better.

You might try the [-Os](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#index-Os) GCC command option.

carlk3, 2024-09-25:
To minimize the size of this library, set these C preprocessor macros to 0:

- USE_PRINTF
- USE_DBG_PRINTF
- SD_CRC_ENABLED

and set this to 1:

- NDEBUG

hellcats88, 2024-09-25:
These are the results of the new compilation, for me now is better and I thinks that will be sufficient. 

WITHOUT f_mount invocation:
```
add_compile_options(-ffunction-sections -fdata-sections)
target_link_options(candle PUBLIC "-Wl,--gc-sections")
add_compile_options(-Os)

Memory region         Used Size  Region Size  %age Used
           FLASH:       69096 B         2 MB      3.29%
             RAM:        256 KB       256 KB    100.00%
       SCRATCH_X:          0 GB         4 KB      0.00%
       SCRATCH_Y:          0 GB         4 KB      0.00%
```
------------------------------------------------------------------------
WITH f_mount invocation:
```
add_compile_options(-ffunction-sections -fdata-sections)
target_link_options(candle PUBLIC "-Wl,--gc-sections")

Memory region         Used Size  Region Size  %age Used
           FLASH:      127156 B         2 MB      6.06%
             RAM:        256 KB       256 KB    100.00%
       SCRATCH_X:          0 GB         4 KB      0.00%
       SCRATCH_Y:          0 GB         4 KB      0.00%
```
```
add_compile_options(-ffunction-sections -fdata-sections)
target_link_options(candle PUBLIC "-Wl,--gc-sections")
add_compile_options(-Os)

Memory region         Used Size  Region Size  %age Used
           FLASH:      111696 B         2 MB      5.33%
             RAM:        256 KB       256 KB    100.00%
       SCRATCH_X:          0 GB         4 KB      0.00%
       SCRATCH_Y:          0 GB         4 KB      0.00%
```

```
add_compile_options(-ffunction-sections -fdata-sections)
target_link_options(candle PUBLIC "-Wl,--gc-sections")
add_compile_options(-Os)
add_compile_definitions(USE_PRINTF=0)
add_compile_definitions(USE_DBG_PRINTF=0)
add_compile_definitions(SD_CRC_ENABLED=0)
add_compile_definitions(NDEBUG=1)

Memory region         Used Size  Region Size  %age Used
           FLASH:      106376 B         2 MB      5.07%
             RAM:        256 KB       256 KB    100.00%
       SCRATCH_X:          0 GB         4 KB      0.00%
       SCRATCH_Y:          0 GB         4 KB      0.00%
```

closed 2024-09-25 by hellcats88