carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #68: Enable support for SDSC cards... again

diegoparrilla, 2024-09-24:
Hi there!

I recently discovered that version 2.5.0 no longer supports SDSC cards, much to my surprise. I'm using your fantastic library in my project: https://github.com/sidecartridge/atarist-sidecart-raspberry-pico.

We have over 1000 devices connected to the classic Atari ST, and quite a few users still rely on SDSC cards.

I recently released new firmware using version 2.6.0, not realizing that SDSC support had been dropped. Now, I’ll need to roll back to v2.4.0 and push out a firmware update.

Is there any way to re-enable SDSC support, either through configuration or programmatically? I’d love to know if there’s a flag or setting that could bring it back.

Thank you so much, and keep up the great work!

carlk3, 2024-09-24:
My problem is that I don't have an SDSC card to test with, and I don't know where to find one. Optimizations have made the logic around Multiple Block reads and writes more complex, and I don't know how much of that will still work with the old, simple `sector *= 512`  conversion.

As the old saying goes, "If it ain't tested, it don't work!"

diegoparrilla, 2024-09-24:
Fair enough. To be honest, I also don't have a single SDSC card for testing.

I have rolled back to a previous version of the library, and I will inform the users that we will drop support for SDSC in the coming releases.

closed 2024-09-24 by diegoparrilla

matsobdev, 2024-11-01:
I guess there is plenty of drivers, PLCs and stuff designed back when SDSC was an only option, like trains. Some modern has drivers requiring 2 GB max card for an update. Mouser, DigiKey and places like that. They have it. They are not cheap, (for one GB) but I guess there is still place in the market for them. And some of them in Poland on an auction site, they sell them like an old stock for some crazy prices :D