carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #69: Example of  SDIO card with Arduino TinyUSB ?

ifengchao, 2024-09-26:
Hello, is it possible to add such an example, useful for projects to read/write files to board from PC.
I wonder what speed can be, USB maight be the bottle neck.
I searched around, no such example availble yet.

carlk3, 2024-09-26:
I'm not sure what you mean. Could you provide a "use case"?

ifengchao, 2024-09-27:
Thank you for your quick reply, let me explain, SDIO interface connect to RP2040, and then USB on board connect to Windows or Mac.
Just like a USB Card reader, useful to export log or upload files to the card without pluging the card off.

Here's the example for rpi pico, but seems only support SPI interface with low speed.
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/MassStorage

carlk3, 2024-09-27:
I might be able to put something like that together, but for Arduino you should probably start with [SdFat](https://github.com/greiman/SdFat).

ifengchao, 2024-09-27:
> I might be able to put something like that together, but for Arduino you should probably start with [SdFat](https://github.com/greiman/SdFat).

That would be awesome! Looking forwarding to this!
I read tinyusb, SdFat, and your fast SDIO lib, but didn't figure how can they combine together, thanks again for your time.

udo-munk, 2024-09-27:
Implemented in the various z80pack machines, there you can export the SD card as disk drive to a PC running Windows, OSX/MacOS, Linux. With this one doesn't need to take out to SD card and put it into some PC with an adapter anymore.

For this Raspberry Pi provides stdio_msc_usb, we had problems under Windows with their implementation.  For testing have a look at this project, this one is easy to reproduce and includes the fixed stdio_msc_usb:
https://github.com/udo-munk/RP2040-GEEK-80

diegoparrilla, 2024-09-27:
In the SidecarTridge Multi-device, I've implemented a feature called "Mass-storage mode," which allows the device to function as an external USB drive that can be mounted on any PC, Linux, or Mac system. This makes transferring files between the Atari ST and a desktop computer much easier. It sounds similar to what you're looking to implement.

Since my device uses SPI from a modern computer, it's quite slow—but still fast enough for the retro systems. If I had more GPIOs available, I’d definitely go with SDIO for faster speeds.

The code here: https://github.com/sidecartridge/atarist-sidecart-raspberry-pico/blob/main/romemul/usb_mass.c
And a silly video showcasing the feature here: https://youtu.be/h6nc69UldEw?si=qH4otj22vXkclLXq

carlk3, 2024-09-27:
I added  `examples/usb_mass_storage` in the [dev](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev) branch.

carlk3, 2024-09-27:
> For this Raspberry Pi provides stdio_msc_usb, we had problems under Windows with their implementation. For testing have a look at this project, this one is easy to reproduce and includes the fixed stdio_msc_usb: https://github.com/udo-munk/RP2040-GEEK-80

I'll have to take a look at that. My implementation seems to run slowly. I wonder if the z80pack fixes make it any faster. I've only tested it on Linux so far, so I haven't come across any problems under Windows.

carlk3, 2024-09-28:
I ran some quick tests with [iozone](https://www.iozone.org/) and compared with an ANKER SD card adapter dongle:
`iozone -a`:

Pico 2, SDIO interface at 25 MHz:
```
	Run began: Fri Sep 27 21:23:20 2024

	Auto Mode
	Command line used: iozone -a .
	Output is in kBytes/sec
	Time Resolution = 0.000001 seconds.
	Processor cache size set to 1024 kBytes.
	Processor cache line size set to 32 bytes.
	File stride size set to 17 * record size.
                                                                    random    random      bkwd     record     stride                                        
              kB  reclen    write    rewrite      read    reread      read     write      read    rewrite       read    fwrite  frewrite     fread   freread
              64       4    369844    359444   2067979   2379626   2067979    304761   1363961     415659    1828508    538156    302018   1279447   1210227
              64       8    426897    392561   2298136   1780008   5283570    598110   2801873     463774    3363612    404998    718114   2006158   3203069
              64      16    138240    703068   3958892   7940539   5735102    688639   2278628     913649    4564786    725882    577521   3363612   4274062
              64      32    200653    799377   5389653   3738358   6421025    444572   2133730    1232453    3022727    666414   1124074   3363612   4274062
              64      64    227718    582535   9006179   9006179   4988978    610350   3022727     780776    3791156    983980    829001   5283570   4018152
...
            2048      32    448050    945007   3717289   4221501   6149699    879773   3374151    3047366    3215029    856184   1042239   3020577   4265523
            2048      64    414816   1132070   5238385   6149699   7112016    917943   4405533    3312988    3944307   1113579   1072829   5044577   4751580
            2048     128    442624   1162403   6401772   8190205   7810411   1009416   4738475    1924783    4223576   1534244   1319740   5225638   4531020
            2048     256    468990   1335124   5489457   5868220   3946119    987484   5264067    2691246    7213548   1339497   1265122   5852228   3074635
            2048     512    489711   1299574   4196751   6024617   7336772   1070022   5006356    2311728    6037320   1584915   1512630   4521480   4405533
            2048    1024    494788   1327285   4853603   4676561   8190205   1218291   4796687    2011785    5056455   1684365   1749190   4138120   5640860
            2048    2048    505921   1151187   7962448   3908414   7556202   1321161   4531020    1639674    3677503   1004458   1199744   5640860   5685664
```
and the dongle:
```
                                                                    random    random      bkwd     record     stride                                        
              kB  reclen    write    rewrite      read    reread      read     write      read    rewrite       read    fwrite  frewrite     fread   freread
              64       4    195680    277078   2222043   2052169   1421755    243190    801764     686877    1336792    516414    294723   1279447   2379626
              64       8    139388    645579   1933893   2133730   1143223    397208   1421755     627470    2067979    633392    432398   2801873   3738358
              64      16    192730    429630   3022727   1690338   2133730    474430   1768283    1304314    4274062    666414    645579   2467108   4274062
              64      32    288390    555990   3203069   2561267   3541098    481234   3022727     955947    5389653    818885    679917   1933893   5283570
              64      64    214610    426897   4274062   5735102   3203069    645579   1947927     743988    3541098    901377    489126   2561267   4018152
...
            2048       4    306544    527546   1715309   2709070   2282853    487266   1314489     895922    1882599    621926    561110   1705432   2625441
            2048       8    358424    698493   2158380   2832339   2778290    709337   1556202    1541125    3012103    748525    751867   2174223   3140967
            2048      16    424030    845565   2951050   4122233   4229816   1007285   2734946    2298737    3764533   1090535    977038   2934917   3118163
            2048      32    439949   1045919   5119743   4347560   6114678   1050781   3512108    2447410    2649737   1175607    953504   3894239   4983122
            2048      64    415920    976594   4829046   4284671   4911886   1075650   3592903    3632406    4081105    901091   1128946   4164199   5970183
            2048     128    436862   1106407   4221501   4540600   6459541   1341170   3536690    2782790    9308497   1226992    896671   4594025   5433896
            2048     256    440604   1276781   5611381   8643474   8128206   1185831   5416763    3506373    6363830   1262704   1366343   5107566   5920802
            2048     512    458281   1432179   5238385   4686767   5900467   1163505   3977179    2283460    4730646   1484143   1447382   3878414   5582208
            2048    1024    464501   1107548   5704543   7503399   8713618   1063267   3764533    1647852    5820505   1457698   1802788   4441984   5264067
            2048    2048    527838   1474462   8498106   8531869   9024882   1491617   3567536    1622945    4414590   1615315   1666393   5251194   5107566

```

udo-munk, 2024-09-28:
It worked fine with Linux and OSX and MacOS. Windows 11 became unresponsible after connecting the USB mass storage device and after a few minutes blue screen. Thomas Eberhardt figured out what the problem is and fixed it.

The performace of the USB drives when using SDIO is quite good, comparable to the usual USB stick drives, so I didn't bother to measure it with iozone and the like.

With SPI and .baud_rate = 125 * 1000 * 1000 / 4, /* 31.25 MHz */ it works, but often times out. Reading files from the SD is not a problem, but writing often causes filesystem corruption because timed out writes.

ifengchao, 2024-09-28:
> I added `examples/usb_mass_storage` in the [dev](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev) branch.

Thank you, very fast coding speed! By the way, is it possible to add a PlatformIO example like [this one](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev/examples/PlatformIO/one_SDIO), I use the PlatformIO and [Adafruit_TinyUSB_Arduino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino), I have this SDK working at hand, Arduino core seems easier for more people to use.

GeekMagicClock, 2024-10-21:
Sorry, this is not a push, I was wondering if it would be possible to share a PlatformIO arduino example of MSC, in the near future?

carlk3, 2024-10-22:
No, not in the near future.