carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #70: Crash when overclock to 270MHz

qdarknight, 2024-09-28:
Hi!
I'm using version v2.6.0. I try to read a file continuously using SDIO.
When overclocking to 250MHz, it's ok.
When overclocking to 270MHz, it seems to crash after several minute(the serial port can't recv data any more).

Code modified from the simple_sdio example:
```
#include <stdio.h>
//
#include "pico/stdlib.h"
//
#include "hw_config.h"
#include "f_util.h"
#include "ff.h"

#include "hw_config.h"

/* SDIO Interface */
static sd_sdio_if_t sdio_if = {
    .CMD_gpio = 2,
    .D0_gpio = 3,
    .baud_rate = 15 * 1000 * 1000  // 15 MHz
};

/* Hardware Configuration of the SD Card socket "object" */
static sd_card_t sd_card = {
    .type = SD_IF_SDIO,
    .sdio_if_p = &sdio_if
};

/* Callbacks used by the library: */
size_t sd_get_num() { return 1; }

sd_card_t *sd_get_by_num(size_t num) {
    if (0 == num)
        return &sd_card;
    else
        return NULL;
}

//#define OVERCLOCK 125000
//#define OVERCLOCK 250000
#define OVERCLOCK 270000

void overclock() {
    set_sys_clock_khz(OVERCLOCK, 1);
}

int min(int x, int y) {
    return x < y ? x : y;
}

int test(char *file) {
    printf("file: %s\n", file);
    FIL fil;
    FRESULT fr = f_open(&fil, file, FA_READ);
    if (fr != FR_OK) {
        printf("f_open fail: %s %d %s\n", file, fr, FRESULT_str(fr));
        return -1;
    }
    int flen = f_size(&fil);
    printf("file length: %d\n", flen);

    int i = 0;
    while (i < flen) {
        char x[1000];
        int l = min(sizeof(x), flen - i);
        int j = 0;
        fr = f_read(&fil, x, l, &j);
        if(fr != FR_OK)
        {
            printf("f_read fail: %d %s\n", fr, FRESULT_str(fr));
            break;
        }
        if(j < l) {
            printf("f_read fail\n");
            break;
        }
        i += l;
        //printf("file offset: %d\n", i);
    }

    fr = f_close(&fil);
    if (fr != FR_OK) {
        printf("f_close fail: %s %d %s\n", file, fr, FRESULT_str(fr));
        return -1;
    }
    if (i < flen) {
        return -1;
    }
    return 0;
}

int main() {
    overclock();

    stdio_init_all();

    puts("Hello, world!");

    // See FatFs - Generic FAT Filesystem Module, "Application Interface",
    // http://elm-chan.org/fsw/ff/00index_e.html
    FATFS fs;
    FRESULT fr = f_mount(&fs, "", 1);
    if (FR_OK != fr) panic("f_mount error: %s (%d)\n", FRESULT_str(fr), fr);
    while(1) {
        test("/a.mp3");
    }
    f_unmount("");

    puts("Goodbye, world!");
    for (;;);
}
```

carlk3, 2024-09-28:
You're on your own there; the maximum core clock speed on RP2040 is 133MHz and on RP2350 is 150MHz.

That said, I recommend choosing the `sdio_if` `baud_rate` to be an integer quotient of `clk_sys`. `sm_config_set_clkdiv` sets the state machine clock divider (from a floating point value) in a PIO state machine configuration. It is a fractional divider, and the jitter introduced by a fractional divisor may be unacceptable. See the data sheet for more.

Do you get any messages if you set the `USE_DBG_PRINTF` C preprocessor macro? Can you attach a debugger to see where you're getting stuck?

qdarknight, 2024-09-29:
Setting clk_sys = 270000000, baud_rate = 22500000, the result is the same.
Setting the USE_DBG_PRINTF macro, there's no difference.

Using gdb to debug, when overclocking to 250MHz, it shows:
```
(gdb) target remote:3333
load
break -source main.c -line 125
continue
Remote debugging using :3333
main () at E:/test/rp2040/pico-sdcard11/main.c:127
127         stdio_init_all();
(gdb) Loading section .boot2, size 0x100 lma 0x10000000
Loading section .text, size 0xfed8 lma 0x10000100
Loading section .rodata, size 0x2f84 lma 0x1000ffd8
Loading section .binary_info, size 0x2c lma 0x10012f5c
Loading section .data, size 0x370c lma 0x10012f88
Start address 0x100001e8, load size 91796
Transfer rate: 23 KB/sec, 10199 bytes/write.
(gdb) Breakpoint 1 at 0x10000450: file E:/test/rp2040/pico-sdcard11/main.c, line 127.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) Continuing.

Breakpoint 1, main () at E:/test/rp2040/pico-sdcard11/main.c:127
127         stdio_init_all();
(gdb)
```
When overclocking to 270MHz, it shows:
```
(gdb) target remote:3333
load
break -source main.c -line 125
continue
Remote debugging using :3333
main () at E:/test/rp2040/pico-sdcard11/main.c:127
127         stdio_init_all();
(gdb) Loading section .boot2, size 0x100 lma 0x10000000
Loading section .text, size 0xfed8 lma 0x10000100
Loading section .rodata, size 0x2f84 lma 0x1000ffd8
Loading section .binary_info, size 0x2c lma 0x10012f5c
Loading section .data, size 0x370c lma 0x10012f88
Start address 0x100001e8, load size 91796
Transfer rate: 23 KB/sec, 10199 bytes/write.
(gdb) Breakpoint 1 at 0x10000450: file E:/test/rp2040/pico-sdcard11/main.c, line 127.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
reset ()
    at E:/test/rp2040/pico-sdcard11/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico-2.6.0/src/src/crash.c:37
37              __breakpoint();
(gdb)
```

BTW, overclocking to 270MHz is not necessary for me.

qdarknight, 2024-09-29:
It's my problem, I forget to set the voltage to 1.2v.
```
void overclock() {
#if OVERCLOCK >= 270000
    /* Above this speed needs increased voltage */
    vreg_set_voltage(VREG_VOLTAGE_1_20);
    sleep_ms(1000);
#endif

    set_sys_clock_khz(OVERCLOCK, 1);
}
```
Then everything is ok.
The code above was copied from the internet, but I don't remember where it came from.

> By default we overclock the RP2040 chip to 250MHz (up from its stock 133MHz) and apply a small over voltage to 1.2v (up from 1.1v), ensuring that it's stable at the higher speed.
https://wiki.picosystem.com/about/hardware

carlk3, 2024-09-30:
Interesting. Glad you got it working! 

Side note: for a few dollars more you could get a [Milk-V Duo](https://milkv.io/duo) and run at 1 GHz with no need to overclock. RISC-V and ARM, kinda like the RP2350.

qdarknight, 2024-09-30:
TY, I'll go learn more about it.

closed 2024-09-30 by qdarknight