carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #72: Building for RISC-V ?

udo-munk, 2024-10-02:
I'm trying to build one of my z80pack VM's for the RISC-V instead of the ARM cores. Looks like FatFs is not prepared for this yet?

Scanning dependencies of target picosim
[ 10%] Building C object CMakeFiles/picosim.dir/picosim.c.obj
In file included from /home/udo/work/z80pack/picosim/rp2350/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/src/sd_driver/sd_regs.h:4,
                 from /home/udo/work/z80pack/picosim/rp2350/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/src/sd_driver/sd_card.h:36,
                 from /home/udo/work/z80pack/picosim/rp2350/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/src/include/hw_config.h:18,
                 from /home/udo/work/z80pack/picosim/rp2350/picosim.c:30:
/home/udo/work/z80pack/picosim/rp2350/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/src/include/util.h:24:10: fatal error: RP2350.h: No such file or directory
   24 | #include "RP2350.h"

carlk3, 2024-10-05:
Right. I haven't even begun to look at RISC-V yet, and it might be a while before I have time. However, if someone else wants to take a crack at porting it, you are welcome to submit a Pull Request.

udo-munk, 2024-10-07:
Understandable. I had a brief look, that is the part the deals with interrupts, DMA and all the low level stuff, must be relearned for RISCV-V, if one isn't familar with the architecture already. Certainly not a quick over the weekend project.

udo-munk, 2024-12-03:
Quick update, we got z80pack working on RISC-V cores and it includes FatOS v3.3.1. Only tested with SPI so far, because I have wired my proto dev board for that.

If you want to have a look:
https://github.com/udo-munk/z80pack/tree/master/picosim/rp2350

udo-munk, 2024-12-04:
Now also tested with a Pico 2 plugged into a Waveshare Pico Eval Board, this is wired for SDIO. VM with disk access works fine on the RISC-V cores.

sneakywumpus, 2024-12-05:
I cleaned the changes I did for z80pack  and put them in #79

udo-munk, 2024-12-08:
Took over v3.7.0 in various z80pack projects and tested on ARM as well as on RISC-V. Works OK, no issues so far.

closed 2024-12-08 by udo-munk