carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #73: SdFat with SDIO support.

greiman, 2024-10-25:
This is not an issue, Just to let you know I finally posted a beta version of SdFat with RP2040/RP2350 support [here](https://github.com/greiman/SdFat-beta).

My main goal was Arduino users with small transfers.  Here are some results from the bench example.

Pico 2 small 64 byte transfers 150 MHz:

> FILE_SIZE_MB = 100
> BUF_SIZE = 64 bytes
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 9980.04,11721,2,6
> 9984.03,8736,2,6
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 10145.07,75,1,5
> 10145.07,76,1,5

Here is the result with 512 byte transfers that avoid cache copies:

Pico 2 512 byte transfers at 150 MHz

> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 15526.96,38,32,32
> 15478.89,39,32,32
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 15526.96,449,32,32
> 15526.96,452,32,32

Finally large transfers and overclocking.

Pico 2 large transfers at 250 MHz

> FILE_SIZE_MB = 100
> BUF_SIZE = 32768 bytes
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 27263.48,11293,1197,1201
> 27256.04,11098,1197,1201
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 27013.02,2400,1210,1212
> 26788.63,9307,1210,1222
>

carlk3, 2024-10-27:
Great! Good news for the Arduino / PlatformIO users. I've retrofitted Arduino compatibility to this library, but prior to that I had never used Arduino, so it was an afterthought.

greiman, 2024-10-27:
> I had never used Arduino

Arduino is not for serious projects.  I have tried  to isolate the core of SdFat so I can use it on systems like ChibiOS.  I see you also have used STM32 chips also.

I had hope for Arduino in 2008 when I first tried it but it never evolved as I expected.  The Arduino guys make lots of money catering to to their huge crowd so it makes good business sense.

greiman, 2024-10-28:
> I've been thinking that 1-bit SDIO

I did a PIO SPI that I could clock at high and got the same rates as 1-bit SDIO and it is really simple.

I will look for the code.

greiman, 2024-10-29:
Here is an "as is" copy of my PIO SPI tests for SdFat for RP2040 from about a year ago.  Very rough - in development.

The zip has the library and a version of bench setup for a Adafruit Metro 2040.

[PioSpi.zip](https://github.com/user-attachments/files/17557665/PioSpi.zip)

I ran it today with this result:

> AdaFruit Metro RP2040 at 250 MHz
> 
> FreeStack: 253840
> Type is exFAT
> Card size: 125.67 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0XAD
> OEM ID: LS
> Product: LX128
> Revision: 2.1
> Serial number: 0X7009FE94
> Manufacturing date: 4/2024
> 
> FILE_SIZE_MB = 20
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 7248.91,87,70,70
> 7251.54,73,70,70
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 7339.36,82,69,69
> 7339.36,745,69,69