carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #76: How to implement SDIO using pins greater than 32

Octopus1633, 2024-11-16:
Hello, can SDIO be implemented using GPIO 34 to 39 of RB2350B? How to modify the program? The connection between RP2350B and SD card is as follows:
```
GPIO34 -- SDIO SCK
GPIO35 -- SDIO CMD
GPIO36 -- SDIO D0
GPIO37 -- SDIO D1
GPIO38 -- SDIO D2
GPIO39 -- SDIO D3
```
I tried to modify SDIO_CLK_PIN_D0_OFFSET
```
.define PUBLIC SDIO_CLK_PIN_D0_OFFSET 18
```

Modify hw_config.c:
```
{   // sdio_ifs[0]
    .CMD_gpio = 35,
    .D0_gpio = 36,
    .CLK_gpio_drive_strength = GPIO_DRIVE_STRENGTH_12MA,
    .CMD_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D0_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D1_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D2_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D3_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .SDIO_PIO = pio1,
    .DMA_IRQ_num = DMA_IRQ_1,
    .baud_rate = 125 * 1000 * 1000 / 8  // 15625000 Hz
    // .baud_rate = 125 * 1000 * 1000 / 7  // 17857143 Hz
    // .baud_rate = 125 * 1000 * 1000 / 6  // 20833333 Hz
    // .baud_rate = 125 * 1000 * 1000 / 5  // 25000000 Hz
    // .baud_rate = 125 * 1000 * 1000 / 4  // 31250000 Hz
},
```

Modify sd_card_sdio.c:
```
    sd_card_p->sdio_if_p->CLK_gpio = (sd_card_p->sdio_if_p->D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32 + 12;
    sd_card_p->sdio_if_p->D1_gpio = sd_card_p->sdio_if_p->D0_gpio + 1;
    sd_card_p->sdio_if_p->D2_gpio = sd_card_p->sdio_if_p->D0_gpio + 2;
    sd_card_p->sdio_if_p->D3_gpio = sd_card_p->sdio_if_p->D0_gpio + 3;
```

But still failed to communicate.What could be the problem?
[image]

carlk3, 2024-11-16:
I haven't tried using RP2350B or high-numbered GPIOs, but here are some things to look at:

- In "11.1.1. Changes from RP2040" of [RP2350 Datasheet](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf): "

  > GPIOBASE adds support for more than 32 GPIOs per PIO block. Each PIO block is still limited to 32 GPIOs at a time, but GPIOBASE selects which 32.

  > GPIOBASE: Relocate GPIO 0 (from PIO’s point of view) in the system GPIO
numbering, to access more than 32 GPIOs from PIO. Only the values 0 and 16 are supported (only bit 4 is writable).

  Are you setting GPIOBASE?

- `SDIO_CLK_PIN_D0_OFFSET` sets the difference between the D0 and SCK GPIOs. If D0 is 36 and `SDIO_CLK_PIN_D0_OFFSET` is 18, I think that would make SCK 36 + 18 mod 32, or 22 (plus `GPIOBASE`).

carlk3, 2024-11-17:
> If D0 is 36 and SDIO_CLK_PIN_D0_OFFSET is 18, I think that would put SCK on GPIO 36 + 18 mod 32, or 22 (plus GPIOBASE).

Rethinking that, the only way to address GPIOs 34-39 would be with `GPIOBASE` set to 16. Assuming that is the case, D0 would be 20 (+ 16), so, a `SDIO_CLK_PIN_D0_OFFSET` of 18 would put SCK on GPIO 

  (20 + 18) mod 32 + `GPIOBASE`
  = 6 + 16

or, again, 22.

Octopus1633, 2024-11-19:
I also have a RP2350 product, which uses GPIO18~23 to connect to the SD card. The connection method is as follows:
```
GPIO18 -- SDIO SCK
GPIO19 -- SDIO CMD
GPIO20 -- SDIO D0
GPIO21 -- SDIO D1
GPIO22 -- SDIO D2
GPIO23 -- SDIO D3
```
When pio_set_gpio_base(SDIO_PIO, 0);, the following configuration can achieve communication with the SD card.
Using Absolute GPIO Numbers (SCK=18 ...D3=23)：
```c
//rp2040_sdio.pio
SDIO_CLK_PIN_D0_OFFSET 30

//hw_config.c
{   // sdio_ifs[0]
    .CMD_gpio = 19,
    .D0_gpio = 20,
    .CLK_gpio_drive_strength = GPIO_DRIVE_STRENGTH_12MA,
    .CMD_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D0_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D1_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D2_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D3_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .SDIO_PIO = pio1,
    .DMA_IRQ_num = DMA_IRQ_1,
    .baud_rate = 125 * 1000 * 1000 / 8  // 15625000 Hz
},

//sd_card_sdio.c
sd_card_p->sdio_if_p->CLK_gpio = (sd_card_p->sdio_if_p->D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32;
sd_card_p->sdio_if_p->D1_gpio = sd_card_p->sdio_if_p->D0_gpio + 1;
sd_card_p->sdio_if_p->D2_gpio = sd_card_p->sdio_if_p->D0_gpio + 2;
sd_card_p->sdio_if_p->D3_gpio = sd_card_p->sdio_if_p->D0_gpio + 3;
```

When pio_set_gpio_base(SDIO_PIO, 16);, the following configurations cannot achieve communication with the SD card.
Using Absolute GPIO Numbers (SCK=18 ...D3=23)：
```c
//rp2040_sdio.pio
SDIO_CLK_PIN_D0_OFFSET 30

//hw_config.c
{   // sdio_ifs[0]
    .CMD_gpio = 19,
    .D0_gpio = 20,
    .CLK_gpio_drive_strength = GPIO_DRIVE_STRENGTH_12MA,
    .CMD_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D0_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D1_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D2_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D3_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .SDIO_PIO = pio1,
    .DMA_IRQ_num = DMA_IRQ_1,
    .baud_rate = 125 * 1000 * 1000 / 8  // 15625000 Hz
},

//sd_card_sdio.c
sd_card_p->sdio_if_p->CLK_gpio = (sd_card_p->sdio_if_p->D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32;
sd_card_p->sdio_if_p->D1_gpio = sd_card_p->sdio_if_p->D0_gpio + 1;
sd_card_p->sdio_if_p->D2_gpio = sd_card_p->sdio_if_p->D0_gpio + 2;
sd_card_p->sdio_if_p->D3_gpio = sd_card_p->sdio_if_p->D0_gpio + 3;
```
Using relative GPIO numbers (SCK=18-GPIOBASE=2 ...D3=7):
```c
//rp2040_sdio.pio
SDIO_CLK_PIN_D0_OFFSET 30

//hw_config.c
{   // sdio_ifs[0]
    .CMD_gpio = 3,
    .D0_gpio = 4,
    .CLK_gpio_drive_strength = GPIO_DRIVE_STRENGTH_12MA,
    .CMD_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D0_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D1_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D2_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D3_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .SDIO_PIO = pio1,
    .DMA_IRQ_num = DMA_IRQ_1,
    .baud_rate = 125 * 1000 * 1000 / 8  // 15625000 Hz
},

//sd_card_sdio.c
sd_card_p->sdio_if_p->CLK_gpio = (sd_card_p->sdio_if_p->D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32 - 2;
sd_card_p->sdio_if_p->D1_gpio = sd_card_p->sdio_if_p->D0_gpio + 1;
sd_card_p->sdio_if_p->D2_gpio = sd_card_p->sdio_if_p->D0_gpio + 2;
sd_card_p->sdio_if_p->D3_gpio = sd_card_p->sdio_if_p->D0_gpio + 3;
```

Octopus1633, 2024-11-19:
Can I modify SDIO_CLK_PIN_D0_OFFSET to be greater than 32? I tried to modify it and the compilation gave an error.
```
C:/Users/xxx/Desktop/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/src/sd_driver/SDIO/rp2040_sdio.pio:117.16-17: pin number must be must be >= 0 and <= 31
  117 |     wait 1 pin SDIO_CLK_PIN_D0_OFFSET  [CLKDIV-1]  ; Wait for rising edge and then whole clock cycle
      |                ^~
```

carlk3, 2024-11-19:
In PIO, pin numbers are modulo 32. You can use `pio_set_gpio_base()` to assign the PIO instance to pins 0-31 or 16-47. See "4.1. Hardware APIs" in **Raspberry Pi Pico-series C/C++ SDK**. Either way, the pin numbers are still 0-31; the difference is whether `GPIOBASE` is 0 or 16.

carlk3, 2024-11-19:
> ```c
> sd_card_p->sdio_if_p->CLK_gpio = (sd_card_p->sdio_if_p->D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32;
> ```
That line is going to have to change to take `GPIOBASE` into account. 

Something like:
```c
sd_card_p->sdio_if_p->CLK_gpio = (sd_card_p->sdio_if_p->D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32 
    + pio_get_gpio_base (sd_card_p->sdio_if_p->SDIO_PIO) ;
```

carlk3, 2024-11-19:
Actually, `GPIOBASE` would need to be added in several places. It might be better to modify `rp2040_sdio_init` in `sd_driver/SDIO/rp2040_sdio.c` to subtract `pio_get_gpio_base(sd_card_p->sdio_if_p->SDIO_PIO)` in the `sm_config_set_in_pins` and `sm_config_set_out_pins` statements. Then, the configuration and everything outside of the PIO could use the absolute pin numbers.

khill25, 2025-03-16:
Are there any plans to add this?

khill25, 2025-03-16:
@Octopus1633 did you ever manage to make set gpio base work for this? I had a similar experience and think I must be missing something. If you do have a working implementation, any chance you can post the diff?

Lithimlin, 2025-06-12:
I would also be very interested in information on this.
Due to my board setup, I cannot change the pins to ones that are below 32…

MatzeS, 2025-06-18:
I can confirm this works with minimal modification and Pico SDK 2.1.1.

Effectively, the only thing to do is moving the pin range with `pio_set_gpio_base`. See [hardware API](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_pio).
The SDK pulls a bunch tricks in the background to make the pin mapping configured through the SDK work with absolute pin numbers (not limited to 32). Hence, this library works effectively out of the box as is since the lib configures the input mapping and the pio code only relies on the relative pin arrangement after the input mapping.

However, the mod 32 clock pin restriction needs to be removed, which allows the clock pin 34 number to propagate to the SDK and the SDK does its magic. Of course, the relative arrangement of SDIO pins must still maintained. The example pin configuration at GPIO34 works for me.

Here is a minimal patch, that is obviously not complete, but with these changes in the library I got it to work: 
```
From 9a4cdfe8e7b068457170d39b54da0fd61a263399 Mon Sep 17 00:00:00 2001
From: Matthias Sokolowski <matthias.sokolowski@tu-darmstadt.de>
Date: Wed, 18 Jun 2025 09:24:07 +0200
Subject: [PATCH 1/1] feat/wip: SDIO on GPIOs > 32

---
 src/sd_driver/SDIO/rp2040_sdio.c  | 3 +++
 src/sd_driver/SDIO/sd_card_sdio.c | 1 -
 2 files changed, 3 insertions(+), 1 deletion(-)

diff --git a/src/sd_driver/SDIO/rp2040_sdio.c b/src/sd_driver/SDIO/rp2040_sdio.c
index 19423d8..9102c13 100755
--- a/src/sd_driver/SDIO/rp2040_sdio.c
+++ b/src/sd_driver/SDIO/rp2040_sdio.c
@@ -782,6 +782,7 @@ bool rp2040_sdio_init(sd_card_t *sd_card_p, float clk_div) {
 
     // Load PIO programs
     pio_clear_instruction_memory(SDIO_PIO);
+    pio_set_gpio_base(SDIO_PIO, 16);
 
     // Command & clock state machine
     STATE.pio_cmd_clk_offset = pio_add_program(SDIO_PIO, &sdio_cmd_clk_program);
@@ -790,6 +791,8 @@ bool rp2040_sdio_init(sd_card_t *sd_card_p, float clk_div) {
     sm_config_set_in_pins(&cfg, SDIO_CMD);
     sm_config_set_set_pins(&cfg, SDIO_CMD, 1);
     sm_config_set_jmp_pin(&cfg, SDIO_CMD);
+    // When using SDK2.0.0: sm_config_set_jmp_pin(&cfg, SDIO_CMD - 16); 
+
     sm_config_set_sideset_pins(&cfg, SDIO_CLK);
     sm_config_set_out_shift(&cfg, false, true, 32);
     sm_config_set_in_shift(&cfg, false, true, 32);
diff --git a/src/sd_driver/SDIO/sd_card_sdio.c b/src/sd_driver/SDIO/sd_card_sdio.c
index e137a32..c470047 100644
--- a/src/sd_driver/SDIO/sd_card_sdio.c
+++ b/src/sd_driver/SDIO/sd_card_sdio.c
@@ -647,7 +647,6 @@ void sd_sdio_ctor(sd_card_t *sd_card_p) {
     myASSERT(!sd_card_p->sdio_if_p->D2_gpio);
     myASSERT(!sd_card_p->sdio_if_p->D3_gpio);
 
-    sd_card_p->sdio_if_p->CLK_gpio = (sd_card_p->sdio_if_p->D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32;
     sd_card_p->sdio_if_p->D1_gpio = sd_card_p->sdio_if_p->D0_gpio + 1;
     sd_card_p->sdio_if_p->D2_gpio = sd_card_p->sdio_if_p->D0_gpio + 2;
     sd_card_p->sdio_if_p->D3_gpio = sd_card_p->sdio_if_p->D0_gpio + 3;
-- 
2.49.0
```

Further reading and notes:
I was previously on SDK 2.0.0 which has a bug on the JMP pin mapping, which is not correctly shifted to the higher pin range (I believe).
However, if SDK 2.0.0 is desired, this can be fixed by manually shifting the JMP pin mapping before passing it to the SDK (see commented code in patch above).

Debugging notes:
The SDK makes a bunch of argument checks and exits early with error code if parameters are invalid. As the library drops some of these error codes in context of PIO configuration, during debugging it should be checked, that the SDK succeeds on all calls. Otherwise config registers may not be set correctly.

References:
[Datasheet](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf) Section 11.4 Instruction Set, and Section 11.7 Registers.