carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #77: Should locks be released on f_open errors

cluoma, 2024-11-29:
I have an SD card attached with SPI and was experimenting with what happens when removing the SD card between accessing files. Specifically `f_open` will return with an error but doesn't seem to release locks, causing the program to stall if  `f_open` is called again:

```
fr = f_open(&fil, filename, FA_OPEN_APPEND | FA_WRITE);
if (FR_OK != fr) {
    printf("couldn't open file\r\n");
}
fr = f_open(&fil, filename, FA_OPEN_APPEND | FA_WRITE);
printf("this will never be displayed\r\n");
```

However, the following will work:

```
fr = f_open(&fil, filename, FA_OPEN_APPEND | FA_WRITE);
if (FR_OK != fr) {
    printf("couldn't open file\r\n");
    sd_card_t * sd = sd_get_by_num(0);
    sd_unlock(sd);
    sd_spi_unlock(sd);
}
fr = f_open(&fil, filename, FA_OPEN_APPEND | FA_WRITE);
printf("this will never be displayed\r\n");
```

carlk3, 2024-12-04:
Yes, that was a mistake. I checked in a fix to the [dev](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev) branch; Commit f5f31b3. Please give that a try.

cluoma, 2024-12-06:
I'm not getting any stalling anymore using the dev branch. Thanks!

closed 2024-12-06 by cluoma