carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #78: pico/aon_timer.h no such file or directory

frangipp, 2024-12-04:
Hello, it's my first time trying your library !

It seems really interesting but I can't compile a project.

I'm using the pico C++ SDK (from mars 2023) with cmake on vscode.

I have copy/paste the "no-OS-FatsFS-SD-SDIO-SPI-RPi-Pico" library in my project folder.
my CMakeLists.tkt : 

```
cmake_minimum_required(VERSION 3.13)

#Abs path of the SDK
set(PICO_SDK_PATH "C:/VSARM/sdk/pico/pico-sdk")

#initialize the SDK based on PICO_SDK_PATH
#note: this must happen before project()
include(pico_sdk_import.cmake)

project(complex_2D_trajectory)

add_compile_options(-Ofast)

#initialize the Raspberry Pi Pico SDK
pico_sdk_init()

#====================

add_subdirectory(
	libs/SD/src build	
)

add_executable(main
    main.cpp
	quick_libs/logs.cpp
	quick_libs/lightMaths.cpp
	positioning/positionControl.cpp
	commands/cmd.cpp
)

#Add pio quadrature_encoder program
pico_generate_pio_header(main ${CMAKE_CURRENT_LIST_DIR}/quadrature_encoder.pio)

#Add pico_stdlib library which aggregates commonly used features
target_link_libraries(main
	pico_stdlib
	pico_i2c_slave
	pico_multicore
	pico_aon_timer
	hardware_adc
	hardware_dma
	hardware_pwm
	hardware_i2c
	hardware_pio
	hardware_clocks
	hardware_exception
	no-OS-FatFS-SD-SDIO-SPI-RPi-Pico
)

#enable usb output, disable uart output
pico_enable_stdio_usb(main 1)
pico_enable_stdio_uart(main 0)

#create map/bin/hex/uf2 file in addition to ELF.
pico_add_extra_outputs(main)
```

All seems to compile properly but the "my_rtc.c" file (witch need "pico/aon_timer.h") : 
```
control_moteur_vComp/libs/SD/src/src/my_rtc.c:18:10: fatal error: pico/aon_timer.h: No such file or directory
   18 | #include "pico/aon_timer.h"
      |          ^~~~~~~~~~~~~~~~~~
```
     
Thank you,

carlk3, 2024-12-04:
**Raspberry Pi Pico SDK 2** is required since v3.0.0. You could use branch [v2.6.0](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/v2.6.0) if you can't move to SDK 2.

frangipp, 2024-12-05:
To update the SDK for the v2.1 :

I tried to update the SDK by myself, without success.
I tried to import my project with the pico extension (on VScode), without success.

Finally I have made a new project using the pico extension and imported manually my files.
And I can now use the library !

Thank you

closed 2024-12-05 by frangipp