carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #82: Reduce resulting code size

jeroentaverne, 2024-12-12:
Hi! Thanks for the great work! I switched to the latest version of your code, but the resulting binary is getting too big for my project. If possible could you add a configuration header file to enable/disable support for SDIO / SPI / RTC / file stream / crash / debug and perhaps other options? I removed some code by hand now which seems to work, but a cleaner way would be great!

carlk3, 2024-12-12:
How are you using the library? I can't see how you are picking up filestream, unless you are running the stdio_buffering example or something like it. Similarly, how are you picking up the crash handling unless you are running the command_line example or something like it? 

To minimize the size of this library, you can set these C preprocessor macros to 0:

- USE_PRINTF
- USE_DBG_PRINTF
- SD_CRC_ENABLED

and set this to 1:

- NDEBUG

Ensure that you are using compile options `-ffunction-sections -fdata-sections` and link option `--gc-sections`. Consider [compile options](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html) `-Os` or `-Oz`.

I need to look at the initialization code; I think a recent change might have it accidentally picking up about 412 Bytes of unnecessary code if SDIO is not used.

jeroentaverne, 2024-12-12:
Thanks for the guide to get the size down. I will check it out. For my project I have multiple smaller firmware images inside the 2MB FLASH, so size reduction of a few 100/1000 bytes can make a difference. Off topic, if interested: my firmware selection example code, which is also used in the PicoGUS project can be found here: https://github.com/jeroentaverne/pico_multi_firmware