carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #87: Trouble accessing SD Card using RP2350 SPI

mjakuipers, 2025-02-04:
I am a happy user of your library. Accessing the uSD card in you lib version 3.1.0 worked really good with SDK V2.0.0. After migrating to the SDK V2.1.0 and you latest version the access to the SD card is troublesome, error shown is "No disk, or could not put SD card in to SPI idle state. I have to do several unmounts (even if the card was not mounted) until I can access the card", so hardware is fine. When reverting back to old firmware versions all is fine. Behavior on the SPI signals looks different as well. Checked with various SD cards. Using the standard Pico 2 board.
After some further investigation and enabling the debug options I see the following:

> sd_spi_go_low_frequency: Actual frequency: 0
sd_init_medium:1410: No disk, or could not put SD card in to SPI idle state
sd_card_spi_init:1626: Failed to initialize card
sd_spi_go_low_frequency: Actual frequency: 0
sd_init_medium:1410: No disk, or could not put SD card in to SPI idle state
sd_card_spi_init:1626: Failed to initialize card
sd_spi_go_low_frequency: Actual frequency: 986895
V2-Version Card
R3/R7: 0x1aa
R3/R7: 0x40ff8000
R3/R7: 0xc0ff8000
Card Initialized: High Capacity Card
SD card initialized
sd_spi_go_high_frequency: Actual frequency: 10066329

The actual SPI frequency for some reason cannot be programmed. But why? It is a multi-core application, but SPI is used only in one core. The other SPI interface is used for controlling FRAM, but totally outside FatFS.
It is programmed correctly (I modified the default frequency just to test) after a few unmounts and then the card works.

Any clues?

carlk3, 2025-02-04:
No idea.

Are you messing with the system clock? [spi_set_baudrate](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_spi_1ga37f4c04ce4165ac8c129226336a0b66c) depends on [clock_get_hz](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_clocks_1gaa2e43d722ca3375f53cb65513275a625) working properly.

This looks like something random, like stack corruption. If you're not already, check for problems with compiler [options](https://gcc.gnu.org/onlinedocs/gcc/Option-Summary.html) `-Wall ` and `-fanalyzer`. Use SDK Configuration Defines like `PARAM_ASSERTIONS_ENABLE_ALL=1`, `PICO_MALLOC_PANIC=1`, and `PICO_USE_STACK_GUARDS=1`. (See section _5.1. Full List of SDK Configuration Defines_ in [Raspberry Pi Pico-series C/C++ SDK](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf).)

If none of that helps, try a divide and conquer strategy. Disable or remove large chunks of code to isolate the problem. Narrow it down to the smallest test case that can reproduce the problem.

mjakuipers, 2025-02-05:
Thanks you for the quick reply. Indeed looks to be related with the clock settings, but no idea why. Will look into this further

closed 2025-02-10 by mjakuipers