carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #89: newbie problem - the physical drive cannot work

lafar6502, 2025-03-02:
Hi, i'm trying to use sdcard reader module with pico w and while some test code in micropython seems to work, i'm unable to repeat the success with c++ project.
relevant fragments of my code are below - the main problem is that mount will not succeed. No information is provided on why, just the message
`f_mount error: The physical drive cannot work (3)`

can you point me to some resources about how to diagnose the problem?
For sure the card reader can work with rpi pico because i'm able to get it to read/write files with micropython.

using this big sd card module
[image]

```
void test_simplefs() 
{
    // See FatFs - Generic FAT Filesystem Module, "Application Interface",
    // http://elm-chan.org/fsw/ff/00index_e.html
    FATFS fs;
    FRESULT fr = f_mount(&fs, "", 1);
    if (FR_OK != fr) {
        printf("f_mount error: %s (%d)\n", FRESULT_str(fr), fr);
        return;
    }

    // Open a file and write to it
    FIL fil;
    const char* const filename = "filename.txt";
    fr = f_open(&fil, filename, FA_OPEN_APPEND | FA_WRITE);
    if (FR_OK != fr && FR_EXIST != fr) {
        printf("f_open(%s) error: %s (%d)\n", filename, FRESULT_str(fr), fr);
        return;
    }
    if (f_printf(&fil, "Hello, world!\n") < 0) {
        printf("f_printf failed\n");
    }

    // Close the file
    fr = f_close(&fil);
    if (FR_OK != fr) {
        printf("f_close error: %s (%d)\n", FRESULT_str(fr), fr);
    }

    // Unmount the SD card
    fr = f_unmount("");
    if (FR_OK != fr) {
        printf("unmount error: %s (%d)\n", FRESULT_str(fr), fr);
    }

}

///----

/* Configuration of hardware SPI object */
static spi_t spi = {
    .hw_inst = spi0,  // SPI component
    .sck_gpio = MP_SPI_SCK,    // 18
    .mosi_gpio = MP_SPI_TX, // 19
    .miso_gpio = MP_SPI_RX, //16
    .baud_rate =  125 * 1000 * 1000 / 8  
};

/* SPI Interface */
static sd_spi_if_t spi_if = {
    .spi = &spi,  // Pointer to the SPI driving this card
    .ss_gpio = MP_SDCARD_CS  // 17
};

/* Configuration of the SD Card socket object */
static sd_card_t sd_card = {
    .type = SD_IF_SPI,
    .spi_if_p = &spi_if  // Pointer to the SPI interface driving this card
};

/* ********************************************************************** */

size_t sd_get_num() { return 1; }

/**
 * @brief Get a pointer to an SD card object by its number.
 *
 * @param[in] num The number of the SD card to get.
 *
 * @return A pointer to the SD card object, or @c NULL if the number is invalid.
 */
sd_card_t *sd_get_by_num(size_t num) {
    if (0 == num) {
        // The number 0 is a valid SD card number.
        // Return a pointer to the sd_card object.
        return &sd_card;
    } else {
        // The number is invalid. Return @c NULL.
        printf("!nullcard\n");
        return NULL;
    }
}
```

the micropython code that 'just works'

```python
import machine
import sdcard
import uos

# Assign chip select (CS) pin (and start it high)
cs = machine.Pin(17, machine.Pin.OUT)

# Intialize SPI peripheral (start with 1 MHz)
spi = machine.SPI(0,
                  baudrate=1000000,
                  #polarity=0,
                  #phase=0,
                  #bits=8,
                  #firstbit=machine.SPI.MSB,
                  sck=machine.Pin(18),
                  mosi=machine.Pin(19),
                  miso=machine.Pin(16))

# Initialize SD card
sd = sdcard.SDCard(spi, cs)

# Mount filesystem
vfs = uos.VfsFat(sd)
uos.mount(vfs, "/sd")

# Create a file and write something to it
with open("/sd/test01.txt", "w") as file:
    file.write("Hello, SD World!\r\n")
    file.write("This is a test\r\n")

# Open the file we just created and read from it
with open("/sd/test01.txt", "r") as file:
    data = file.read()
    print(data)

```

lafar6502, 2025-03-02:
update: enabled debug print options and my output is now
so, it looks like i'll need bigger sd card?
```
write test
sd_spi_go_low_frequency: Actual frequency: 398089
sd_spi_go_low_frequency: Actual frequency: 398089
V2-Version Card
R3/R7: 0x1aa
R3/R7: 0x80ff8000
R3/R7: 0x80ff8000
Card Initialized: Standard Capacity Card: Version 2.x
sd_card_spi_init:1632: SD Standard Capacity Memory Card unsupported
f_mount error: The physical drive cannot work (3)
done

```

lafar6502, 2025-03-02:
another update
after switching to 16gb sd card the example started working, file is written. Hooray!
then tried to share the SPI bus with MAX6675
i got it to work, but with some hassle, sometimes these two devices interfere and sd card can't be mounted
has something to do with initialization order and CS pin values. It works when setting max6675 cs low before accessing the sd card, but only on second attempt. I will dig into this further but if there's sometihing obvious i'm missing - let me know.
R

carlk3, 2025-03-02:
That SD card module looks suspiciously like an Arduino module. Those are designed for 5 V signalling instead of 3.3 V. If that's the case, it might work, but it won't work well. It is possible to modify some of them; see [The 5V Arduino SD modules might work with a simple trick](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/issues/83).

For sharing the bus, see [Cosideration on Multi-slave Configuration](http://elm-chan.org/docs/mmc/mmc_e.html#spibus).

carlk3, 2025-03-03:
You can also [make an SD card breakout from a microSD to SD adapter](https://alexlubbock.com/micro-sd-adapter-esp8266-esp32).

lafar6502, 2025-03-03:
thanks! i should be able to continue from here, its almost working - just more stability is needed.

xayhewalo, 2025-03-25:
@lafar6502 What brand SD card are you using?

I have this exact same problem with a 32GB Micro center SD card and an adapter from Catalex. This hardware works with CircuitPython but not with the simple example from this library.

[image]

[image]

carlk3, 2025-03-26:
Have a look at [The 5V Arduino SD modules might work with a simple trick #83](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/issues/83#issuecomment-1700883170).