carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #90: Not working with mmc card

Anjali8340, 2025-04-02:
Hi,
I am trying to log data into MMC card. This library is working well with exfat sd card card but not working with mmc card. does it support mmc card? if yes then can you provide me an example code of that?

carlk3, 2025-04-02:
I don't know of any reason that it shouldn't work, but it hasn't really been tested. What are the symptoms?

Try turning on `DBG_PRINTF`. (See [Messages](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico#messages).) For example, in `CMakeLists.txt`,
```
add_compile_definitions(USE_PRINTF USE_DBG_PRINTF)
```
Now, do you see any messages? If so, post them here.

Anjali8340, 2025-04-03:
Earlier i was using 128gb sd card and it was working perfectly. Now i want to switch to mmc cards.
 i have copied from your one SDIO example with required modification. i am not using card detect pin.

```
/* 
Copyright 2023 Carl John Kugler III

Licensed under the Apache License, Version 2.0 (the License); you may not use 
this file except in compliance with the License. You may obtain a copy of the 
License at

   http://www.apache.org/licenses/LICENSE-2.0 
Unless required by applicable law or agreed to in writing, software distributed 
under the License is distributed on an AS IS BASIS, WITHOUT WARRANTIES OR 
CONDITIONS OF ANY KIND, either express or implied. See the License for the 
specific language governing permissions and limitations under the License.
*/

/* Write "Hello, world!\n" to SD Card */
#include <string.h>
#include "FatFsSd_C.h"
//
#include "Arduino.h"

#define printf Serial.printf
#define puts Serial.println

/* Implement library message callbacks */
void put_out_error_message(const char *s) {
    Serial.write(s);
}
void put_out_info_message(const char *s) {
    Serial.write(s);
}
// This will not be called unless build_flags include "-D USE_DBG_PRINTF":
// void put_out_debug_message(const char *s) {
//     Serial.write(s);
// }

/*
This example assumes the following wiring:
    | GPIO | SD Card |
    | ---- | ------- |
    | GP2  | CLK     |
    | GP3  | CMD     |
    | GP4  | D0      |
    | GP5  | D1      |
    | GP6  | D2      |
    | GP7  | D3      |
    | GP9  | DET     |
*/
static sd_sdio_if_t sdio_if = {
    /* 
    Pins CLK_gpio, D1_gpio, D2_gpio, and D3_gpio are at offsets from pin D0_gpio.
    The offsets are determined by sd_driver\SDIO\rp2040_sdio.pio.
        CLK_gpio = (D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32;
        As of this writing, SDIO_CLK_PIN_D0_OFFSET is 30, 
            which is -2 in mod32 arithmetic, so:
        CLK_gpio = D0_gpio -2.
        D1_gpio = D0_gpio + 1;
        D2_gpio = D0_gpio + 2;
        D3_gpio = D0_gpio + 3;
    */
    .CMD_gpio = 3,
    .D0_gpio = 4,
    .SDIO_PIO = pio1,
    .DMA_IRQ_num = DMA_IRQ_1,
    .baud_rate = 15 * 1000 * 1000   // 20833333 Hz
};

// Hardware Configuration of the SD Card "objects"
static sd_card_t sd_card = {
    .type = SD_IF_SDIO,
    .sdio_if_p = &sdio_if,
    // SD Card detect:
    .use_card_detect = false,
    // .card_detect_gpio = 8,  
    // .card_detected_true = 0, // What the GPIO read returns when a card is present.
};
/* 
The following functions are required by the library API. 
They are how the library finds out about the configuration.
*/
extern "C" size_t sd_get_num() { return 1; }

extern "C" sd_card_t *sd_get_by_num(size_t num) {
    if (0 == num) {
        return &sd_card;
    } else {
        return NULL;
    }
}

// Check the FRESULT of a library call.
//  (See http://elm-chan.org/fsw/ff/doc/rc.html.)
#define CHK_FRESULT(s, fr)                                  \
    if (FR_OK != fr) {                                      \
        printf("%s:%d %s error: %s (%d)\n",                 \
               __FILE__, __LINE__, s, FRESULT_str(fr), fr); \
        for (;;) __breakpoint();                            \
    }

void setup() {
    Serial.begin(115200);  // set up Serial library at 9600 bps
    while (!Serial.available()) 
        ;  // Serial is via USB; wait for enumeration
    printf("\033[2J\033[H");  // Clear Screen
    puts("Hello, world!");

    // See FatFs - Generic FAT Filesystem Module, "Application Interface",
    // http://elm-chan.org/fsw/ff/00index_e.html
    sd_card_t *pSD = sd_get_by_num(0);
    FRESULT fr = f_mount(&pSD->state.fatfs, "", 1);
    printf("f_mount: %s\n", FRESULT_str(fr));
   // CHK_FRESULT("f_mount", fr);
    FIL fil;
    const char* const filename = "filename.txt";
    fr = f_open(&fil, filename, FA_OPEN_APPEND | FA_WRITE);
    CHK_FRESULT("f_open", fr);
    char const * const str = "Hello, world!\n";
    if (f_printf(&fil, str) < strlen(str)) {
        printf("f_printf failed\n");
        for (;;) __breakpoint();
    }
    fr = f_close(&fil);
    CHK_FRESULT("f_close", fr);
    fr = f_unmount("");
    CHK_FRESULT("f_unmount", fr);

    puts("Goodbye, world!");
}
void loop() {}
```

this is the message on serial terminal

```
Hello, world!
sd_sdio_begin,118 SDIO not responding to CMD8 SEND_IF_COND, status 0x2 reply 0x0
f_mount: The physical drive cannot work
sd_sdio_begin,118 SDIO not responding to CMD8 SEND_IF_COND, status 0x2 reply 0x0
```

Anjali8340, 2025-04-05:
it is not even working after lowering the baudrate upto1MHz. There are some command  difference between emmc and sd card. I think you should check it the commands of emmc.

carlk3, 2025-04-06:
> There are some command difference between emmc and sd card.

You're right. I just downloaded [MultiMediaCard (MMC) Electrical Standard, High Capacity (MMCA, 4.2), JESD84-B42, JULY 2007 ](https://www.jedec.org/document_search?search_api_views_fulltext=JESD84-B42) and am looking at it. Is that the correct standard for the MMC card you have?

Anjali8340, 2025-04-07:
Well i am using eMMC 5.1 card and i feel [JESD84-B50, SEPTEMBER 2013](https://www.jedec.org/sites/default/files/docs/JESD84-B50.pdf) should be correct standard for my card as it says eMMC 5.0 standards. 
Are you going to implement it?

carlk3, 2025-04-09:
I'm looking at it. I got one of these: [EMMC Module 64GB 32GB 16GB 8GB with Micro SD-compatible Turn eMMC Adapter](https://www.aliexpress.us/item/3256804319492196.html) to experiment with. Eventually, I'd like to implement eMMC with 8-bit-wide bus, for a doubling of data transfer speed, and Dual Data Rate (DDR), for another potential doubling in speed.