carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #91: Adding support for a DS1307 RTC

tgiacchi, 2025-04-09:
I'm curious about the best/proper way of adding in a DS1307 rtc into the code.  I have the hardware and code working outside of your code.  Should I just replace the get_fattime call? Or better to initialize the time from the RTC on start up and let your normal code handle the rest?

carlk3, 2025-04-09:
That's a great question, and I have been grappling with a similar problem on another project. 

Three possibilities:

1. Synchronize at startup and just let them drift apart.
2. Synchronize at startup and periodically, say once an hour, thereafter.
3. Query the DS1307, calculate and set `time_t epochtime`  at startup, then use the DS1307's Square Wave/Output Drive to output a square-wave of 1 Hz on SQW/OUT pin, connect that pin to a GPIO on the RP2, and set up `gpio_set_irq_enabled_with_callback` on that GPIO to increment `time_t epochtime` each second. In this case, there would be no need to even start the RP2 RTC or AON timer.

For option 1, it might be worth measuring the delay in the synchronization (1/2 the round trip time?) and compensating for that. Hopefully, you can query the DS1307 while there is no other traffic on the I2C bus.

Presumably, the DS1307 is more accurate than the RP2 internal RTC, and they will drift apart if not periodically synchronized. However, for option 2, if I just randomly query the DS1307 and put that into the RP2 RTC, the DS1307 time could have just changed or could be just about to change, which would cause anomalies in the RP2 RTC time series. To avoid that, you could use some kind of spin loop to try to detect the edge of a DS1307 increment.  If the I2C bus is shared, the synchronization delays might be unpredictable.

To me, option 3 looks like the best solution, but it might require an additional wire and a GPIO. It would give, by far, the most accurate time stamps, and it would be low-overhead.

tgiacchi, 2025-04-09:
No spare gpio.

tgiacchi, 2025-04-09:
How do you get the time? Is everything driven off of the epochtime? Is there a central place where the time is grabbed? And can you clarify the purpose of the get_fattime function ?

tgiacchi, 2025-04-10:
If i take approach 3, I can share the sda line. Config it as gpio edge rising.  Either set a flag and read in main comms loop or switch to i2c read 8 update and switch back. The time_t  is available in one of the fat library structures?

tgiacchi, 2025-04-10:
Doh! Shouldn't design code after surgery.  I don't need to switch i2c at all. Just update the time_t. Sharing the pin still works as I only have one i2c device. Seems like as good thing to move into the main code.

carlk3, 2025-04-10:
Interesting idea! Since the SQW/OUT pin is open drain that could work. If you synchronize the time first, and then enable square-wave output last, you shouldn't need to issue any more I2C commands after the write to the control register. The only thing the interrupt needs to do is to increment a global `epochtime` variable. 

You would need to change `get_fattime` to use `epochtime`:
```
    struct timespec ts;
    bool ok = get_time(&ts);
    if (!ok) return 0;

    struct tm t;
    localtime_r(&ts.tv_sec, &t);
```
to
```
    struct tm t;
    localtime_r(&epochtime, &t);
```

carlk3, 2025-04-10:
> Also have to spin lock around the update as it's 64bit

If you do the `epochtime` increment in the interrupt, what can interrupt you? Only things like Hard Faults, I believe.

> Have to change mode once on pico side so I can get the interrupts. 

Hmmm, yes, presumably the DS1307 is free running, and when you set Square-Wave Enable (SQWE) I guess you can't predict when the first rising edge will occur -- it depends on the DS1307 time. You might miss an interrupt.

carlk3, 2025-04-10:
You could probably tell if you missed an interrupt, though. You could keep a separate "`epochtime` last update" timestamp with a separate timer, e.g., `time_us_64()`. If it has been well over a second since the less update, you can assume you missed an interrupt.

tgiacchi, 2025-04-10:
The issue is if the interrupt happens when getting the value. Low 32bits are grabbed and then the interrupt.  Most times not a problem, however if the low 32bits is 0xfffffff and the interrupt happens, the result would be off by 0xffffffff. Probably a critical section is better.

carlk3, 2025-04-10:
I guess the RP2040 does have four interrupt priority levels, but unless you are running an OS or have set up higher priority interrupts, I don't think you need to worry about an interrupt interrupting an interrupt except for things like Hard Faults. You don't want to be doing a spin lock in an interrupt.

carlk3, 2025-04-10:
> The issue is if the interrupt happens when getting the value. 

Oh, I see what you mean; if the mainline code (Thread mode) is in the middle of reading `epochtime` when the interrupt (Handler mode) happens... Yeah,  maybe `time_t time()` should be `static inline` and it should do a `save_and_disable_interrupts()` and `restore_interrupts()` around the read.

carlk3, 2025-04-10:
Another way would be to simply detect the condition that your have reached the Epochalypse (Y2038) and dropped the carry, and correct for it. I.e., if the previous `epochtime` was `0x00000000FFFFFFFF` and `epochtime` is `0x0000000000000000` change it to `0x0000000100000000`.

tgiacchi, 2025-04-10:
I hate special case logic.  Better to make sure it does not happen.

carlk3, 2025-04-10:
> I hate special case logic.  Better to make sure it does not happen.

I deleted that when I realized that even 32-bit reads and writes are not atomic on Cortex-M0.

So, it's probably a bad idea for `epochtime` to be a global variable. For speed (e.g., for frequent timestamps), it would be nice to make `time_t time()` `static inline`, but that might conflict with the declaration in `<time.h>`, so maybe there could be a `static inline time_t epochtime()` and a standard non-static `time_t time()` that calls `epochtime()`.

carlk3, 2025-04-10:
I stand corrected. Here's how the SDK deals with something similar:
```
uint64_t timer_time_us_64(timer_hw_t *timer) {
    // Need to make sure that the upper 32 bits of the timer
    // don't change, so read that first
    uint32_t hi = timer->timerawh;
    uint32_t lo;
    do {
        // Read the lower 32 bits
        lo = timer->timerawl;
        // Now read the upper 32 bits again and
        // check that it hasn't incremented. If it has loop around
        // and read the lower 32 bits again to get an accurate value
        uint32_t next_hi = timer->timerawh;
        if (hi == next_hi) break;
        hi = next_hi;
    } while (true);
    return ((uint64_t) hi << 32u) | lo;
}
```