carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico #92: Write errors with examples/usb_mass_storage demo

chrisgjohnson, 2025-04-15:
I'm running the new `examples/usb_mass_storage` demo on RP2040 hardware that has previously been working well with this library for read-only SDIO access to an SD card.

When running the MSC demo, the partition mounts fine, and reading data from the SD card seems to work, but when editing/copying files I'm getting frequent write errors (`check_sdio_write_response` reports `SDIO card reports write CRC error`...), which soon leads to file system errors on the SD card. 

This happens on multiple SD cards, all with exFAT partitions. The USB host is is a Linux desktop.

This 'issue' is really to ask whether this could be an issue with the firmware?

Or, if this demo is well tested without the sort of issue occurring, perhaps it points to signal integrity issues in my hardware that haven't manifested in previous uses.

carlk3, 2025-04-16:
The most common cause of CRC errors is trying to run at too high a baud rate for the hardware. You could try cutting the baud rate in half to see if the CRC errors go away.

baud_rate = clk_sys / (CLKDIV * clk_div) where CLKDIV is set to 4 in the PIO code. Preferrably, choose `baud_rate` for an integer clk_div. It shouldn't matter much, but it might give a more symmetrical CLK signal.

See [An instance of sd_sdio_if_t describes the configuration of one SDIO to SD card interface.](https://github.com/carlk3/no-OS-FatFS-SD-SDIO-SPI-RPi-Pico/tree/dev?tab=readme-ov-file#an-instance-of-sd_card_t-describes-the-configuration-of-one-sd-card-socket).

chrisgjohnson, 2025-04-16:
Thanks Carl - it was indeed a baud rate problem. (In this case a non-integer `clk_div` made quite a big difference to the maximum baud rate - I think more so than I've seen before on this hardware, which is what confused me initially). Of course, for USB mass storage, data transfer over USB FS is going to be the limiting factor, so no advantage in pushing the SDIO baud rate.

closed 2025-04-16 by chrisgjohnson